vindar/ILI9341_T4 #1: Does not work with Teensy 4.0 on SPI 0

DIYLAB-DE, 2021-07-12:
Hi,
unfortunately the example "99 balloons" does not work with Teensy 4.0 and an ILI9341 3,2" display at SPI 0.
The screen stays dark (backlight LED is configured correctly).

Did you try your driver only on Teensy 4.1?

Best regards
[99luftballons.zip](https://github.com/vindar/ILI9341_T4/files/6802071/99luftballons.zip)

DIYLAB-DE, 2021-07-12:
Hi,

thanks for the explanations, now it is clear to me.

Ok, then I can not use this driver, too bad.

In my case MISO is not connected and Reset is on VCC:

https://www.diylab.de/forum/thread/31-huckepack-platine-im-entstehen/?postID=162#post162

 

Kind regards

Bruno

vindar, 2021-07-12:
Hi, 

Yes indeed, if MISO is not connected then there is not way to use this library but [ILIT9341_t3](https://github.com/PaulStoffregen/ILI9341_t3) or [ILI9341_t3n](https://github.com/KurtE/ILI9341_t3n) should work nicely.  

nice project btw. 

Best,

closed 2021-07-12 by vindar