vindar/ILI9341_T4 #9: Thanks for the update cs on any pin!

MaltWhiskey, 2022-07-15:
Works like a charm!

https://youtu.be/lhVIrq3rMb4

And thanks for the game of life example :)

May I suggest this small change to keep things interesting?

// Hash list containing the hashes of past generations
uint32_t hash_list[8192];
// Amount of hashes available and position to store new  hash in
uint8_t hash_nr = 0;

void check_world(uint32_t hash) {
  uint16_t matches = 0;
  for (uint16_t i = 0; i < hash_nr; i++)
    if (hash_list[i] == hash) matches++;
  if (matches >= 6 || hash_nr == 8191) {
    hash_nr = 0;
    random_world(oldworld, 0.7f);
  } else {
    hash_list[hash_nr++] = hash;
  }
}

// compute the world at the next generation.
uint32_t compute(uint8_t* cur_world, uint8_t* new_world) {
  uint32_t hash = 0;
  for (int j = 1; j < WORLD_LY - 1; j++) {
    for (int i = 1; i < WORLD_LX - 1; i++) {
      const int ind = i + (WORLD_LX * j);
      const int nbn = nb_neighbours(i, j, cur_world);
      hash += nbn * (j * 3 + i * 5);
      if (nbn == 3)
        new_world[ind] = 1;
      else if ((nbn < 2) || (nbn > 3))
        new_world[ind] = 0;
      else
        new_world[ind] = cur_world[ind];
    }
  }
  return hash;
}

void loop_lcd() {
  uint32_t hash = compute(curworld, oldworld);  // compute the next geenration
  check_world(hash);
  swap_worlds();                       // swap between old and new worlds
  draw_world(curworld, WHITE, BLACK);  // draw onto the framebuffer

vindar, 2022-07-16:
Hi, 

thanks for the improvement in the game of life example ! I will update the example soon :)

vindar, 2022-07-27:
Hi, 

I added you hash checking code to the example. Thank you :-)

closed 2022-07-27 by vindar