vindar/ILI9341_T4 #10: ILI9341_T4Driver: Hanging in _waitUpdateAsyncComplete()...

MaltWhiskey, 2022-07-16:
When I send data over the serial connection (from a external esp8266) the driver will hang.
ILI9341_T4Driver: Hanging in _waitUpdateAsyncComplete()...

This is the setup for serial between teensy and esp
  // ESP8266 UART baudrate on Hardware Serial1
  Serial1.begin(460800);
  // Prevents RX buffer overflow if not reading fast enough
  static char read_buffer[4096];
  Serial1.addMemoryForRead(read_buffer, sizeof(read_buffer));

I have made a yield to keep handeling serial data so the error message will keep showing up in the serial monitor (so the program is still active, just stuck forever in the _wait loop.
// Check custom serial events (yield gets called in ILI9341_T4Driver)
void yield() { ESP8266::loop(); }

The ESP8266::loop doesn't do anything special but read from Serial1 and write this to Serial. If I go for the default yield() nothing is printed in the serial monitor, but the result is still the same.

I put a break after this
if (em > 1000)

but this will crash the teensy.

Any clue at what is making the driver block on _dma_state != ILI9341_T4_DMA_IDLE ??

I haven't investigated any further, thought I'd ask first ;)

Thx
Malt Whiskey

MaltWhiskey, 2022-07-16:
There must be something that messes up your interrupts in the HardwareSerial thereby missing the IDLE flag... https://github.com/PaulStoffregen/cores/blob/master/teensy4/HardwareSerial.cpp

vindar, 2022-07-16:
Hi,

Weird bug. Clearly looks interrupt related. I have not stumbled upon it before... I will try to reproduce and look at it asap but it may take a few days since I just relocated and have yet to unpack my computer/electronic stuffs...

MaltWhiskey, 2022-07-18:
Thanks, if i can help in anyway lmk

(i’m on a vacation next week, no coding/computer allowed)

vindar, 2022-07-20:
Hi, 

Indeed, I can reproduce the bug ! I just added polling of Serial1 to the 99luftballons example (and used another Teensy to send random bytes via Serial1). 

In my test, the bug does not occur when DC is wired to an hardware CS pin of the Teensy:  this hints at a conflict with the SPI interrupt (because those interrupts are used much more intensively when DC is software emulated).

A (not completely satisfying) workaround is to increase the priority of the interrupt for the ILI941_T4 driver.  To do so, just edit line 79 of `ILI9341Driver.h` and set it to:
```
#define ILI9341_T4_IRQ_PRIORITY 64
```

I have not tested this fix exhaustively but at least my test program does not hang anymore after the priority is increased... Yet, I would still like to understand exactly what is happening here :-)

Thank you for reporting the bug !

vindar, 2022-07-21:
Hi again, 

I think I have found the problem: when an interrupt occurs around lines 1487/1488 in `ILI9341Driver.h` and lasts too long, the SPI bus has time to complete the transaction but the interrupt flag is cleared so that the next interrupt is never called...

I just pushed a tentative fix (which does not require to increase the interrupt priorities anymore).  Please let me know it it works for you when you have the time to test it. 

Enjoy your holidays in the meanwhile ;-)

MaltWhiskey, 2022-07-21:
Hi Arvind,

That was quick :-)
I’ll let you know asap when i get home if the bug is fixed.

MaltWhiskey, 2022-07-26:
Tested and it works! Issue solved :)

vindar, 2022-07-26:
Great :-)

Thanks again your help !

closed 2022-07-26 by vindar