vindar/ILI9341_T4 #11: OverlayTextTFTDisplay

ArminRehberger, 2022-09-25:
Hello Arvind,
I use your libary to display the X, Y and Z positions of my programmed CNC mill with the Teensy4.0.
I implemented a new method to draw text on the TFT screen.
It would be nice if you could implemend this method in your libary.

In this new method you can:
Juse the line number
Draw universal text
Juse one of your fonts
Juse the width of the rect for the background of the text (draw rect like textlength, draw rect over the entire width)

I found out an error in your method "FillRect" during testing the new method when drawing textz at the bottom.
//if (ymax >= lx) ymax = ly - 1; // Error
if (ymax >= ly) ymax = ly - 1;

Kind regards
Armin Rehberger

// ##### Draw text
// - fb : the framebuffer to draw onto
// - position: position of the text:
//             0= top right, 1=bottom right, 2=bottom left, 3=top left
// - line: line no.
//            0=line 0, 1=line 1, ...
// - fg_color : foreground color.   
// - bg_color : background color
// - opacity: opacity of the FPS counter between 0.0f=transparent and 1.0f=opaque. 
// - text: Text to draw
// - font: Font to be used:
//         0=Bold_10, 1=Bold_12, 2=Bold_14, 3=Bold_16
// - Draw rect over the entire width
//        0=draw rect like textlength, 1=draw rect over the entire width

// ##### Call the new method
char outStr[100];
sprintf(outStr," ActPos X: %.4f mm",MotorRecive.ActPosMotor[0]); // Actual position X
tft.OverlayTextTFTDisplay(fb, 3, 0, ILI9341_BLACK, ILI9341_RED, 0.8f, outStr, 3, 1); // Framebuffer, position, line, foreground color, background color, opacity, text, font, rect entire width

sprintf(outStr," ActPos Y: %.4f mm",MotorRecive.ActPosMotor[1]); // Actual position y
tft.OverlayTextTFTDisplay(fb, 3, 1, ILI9341_BLACK, ILI9341_GREEN, 0.8f, outStr, 3, 1); // Framebuffer, position, line, foreground color, background color, opacity, text, font, rect entire width

sprintf(outStr," ActPos Z: %.4f mm",MotorRecive.ActPosMotor[2]); // Actual position z
tft.OverlayTextTFTDisplay(fb, 3, 2, ILI9341_BLACK, ILI9341_BLUE, 0.8f, outStr, 3, 1); // Framebuffer, position, line, foreground color, background color, opacity, text, font, rect entire width

sprintf(outStr," RS485: %d",CounterReadGoodRS485); // Counter read good RS485
tft.OverlayTextTFTDisplay(fb, 3, 3, ILI9341_WHITE, ILI9341_BLACK, 0.5f, outStr, 3, 1); // Framebuffer, position, line, foreground color, background color, opacity, text, font, rect entire width

// #################################################################
// ##### Draw text, new method
/*
- fb : the framebuffer to draw onto
- position: position of the text:
            0= top right, 1=bottom right, 2=bottom left, 3=top left
- line: line no.
        0=line 0, 1=line 1, ...
- fg_color : foreground color.   
- bg_color : background color
- opacity: opacity of the FPS counter between 0.0f=transparent and 1.0f=opaque. 
- text: Text to draw
- font: Font to be used:
        0=Bold_10, 1=Bold_12, 2=Bold_14, 3=Bold_16
- Draw rect over the entire width
        0=draw rect like textlength, 1=draw rect over the entire width
*/
FLASHMEM void ILI9341Driver::OverlayTextTFTDisplay(uint16_t* fb, uint16_t position, uint16_t line, uint16_t fg_color, uint16_t bk_color, float opacity, char* text, uint16_t f_font, uint16_t entireWidth)
{

  // ##### Transfer font
  /*
  extern const ILI9341_t3_font_t font_ILI9341_T4_OpenSans_Bold_10;
  extern const ILI9341_t3_font_t font_ILI9341_T4_OpenSans_Bold_12;
  extern const ILI9341_t3_font_t font_ILI9341_T4_OpenSans_Bold_14;
  extern const ILI9341_t3_font_t font_ILI9341_T4_OpenSans_Bold_16;
  */
  const ILI9341_t3_font_t *font;
  switch (f_font)
  {
    case 0:
      font = &font_ILI9341_T4_OpenSans_Bold_10; // Transfer adress of font
      break;
    case 1:
      font = &font_ILI9341_T4_OpenSans_Bold_12; // Transfer adress of font
      break;
    case 2:
      font = &font_ILI9341_T4_OpenSans_Bold_14; // Transfer adress of font
      break;
    case 3:
      font = &font_ILI9341_T4_OpenSans_Bold_16; // Transfer adress of font
      break;
    default:
      font = &font_ILI9341_T4_OpenSans_Bold_16; // Transfer adress of font
      break;
  }

  // ##### Measure text
  int text_off = 0;
  int x = 0;
  int y = 0;
  int _fps_xmin, _fps_xmax, _fps_ymin, _fps_ymax;
  MeasureText(text + text_off, x, y, _fps_xmin, _fps_xmax, _fps_ymin, _fps_ymax, font, false);
  _fps_xmin -= 2;
  _fps_xmax += 2;
  _fps_ymin -= 3;
  _fps_ymax += 2;

  // ##### Position of the text
  // 0=top right, 1=bottom right, 2=bottom left, 3=top left
  int dx, dy;
  switch (position)
  {
    case 0: // 0=top right
      dx = _width - 1 - _fps_xmax;
      dy = -_fps_ymin;
      break;
    case 1: // 1=bottom right
      dx = _width - 1 - _fps_xmax;
      dy = _height - 1 - _fps_ymax;
      break;
     case 2: // 2=bottom left
      dx = -_fps_xmin;
      dy = _height - 1 - _fps_ymax;
      break;
    case 3: // 3=top left
      dx = -_fps_xmin;
      dy = -_fps_ymin;
      break;
    default: // 0= top right
      dx = _width - 1 - _fps_xmax;
      dy = -_fps_ymin;
      break;
  }

  x += dx;
  _fps_xmin += dx;
  _fps_xmax += dx;
  y += dy;
  _fps_ymin += dy;
  _fps_ymax += dy;

  // ##### Line no.: 0, 1, 2, ...
  // Correct y positions
  int lineHeight;
  lineHeight = _fps_ymax - _fps_ymin;
  // Position: 0= top right, 1=bottom right, 2=bottom left, 3=top left
  switch (position)
  {
    case 0: // Position 0=top right
    case 3: // Position 3=top left
      _fps_ymin += line*lineHeight +(line); // Rect y min
      _fps_ymax += line*lineHeight +(line); // Rect y max
      y += line * lineHeight +(line); // Text y
      break;
    case 1: // Position 1=bottom right
    case 2: // Position 2=bottom left
      _fps_ymin -= line*lineHeight +(line); // Rect y min
      _fps_ymax -= line*lineHeight +(line); // Rect y max
      y -= line*lineHeight +(line); // Text y
      break;
    default: // Position top
      _fps_ymin += line*lineHeight +(line); // Rect y min
      _fps_ymax += line*lineHeight +(line); // Rect y max
      y += line * lineHeight +(line); // Text y
      break;
  }

    // ##### Draw rect over the entire width
  if(entireWidth == 1)
    _fps_xmax = _width;

  // ##### Fill rect and draw text
  FillRect(_fps_xmin, _fps_xmax, _fps_ymin, _fps_ymax, _width, _height, _width, fb, bk_color, opacity);
  DrawTextILI(text + text_off, x, y, fg_color, font, false, _width, _height, _width, fb, opacity + 0.3f);
  return;
}

vindar, 2022-09-26:
Hello, 

I agree with you that is useful to have a simple way to draw text onto the screen (for debugging, displaying infos...). So, as per your suggestion, I have created a new method called `overlayText()` adapted from your code above. I hope this suits you. 

Since the library is just a screen driver (and not a full fledged graphics lib), I do not want to add more methods for drawing onto a framebuffer here.  But, if you need more advanced text formatting, I suggest you look into my tgx library which will give you many more options to do that.  

Thanks you for your help and for reporting the bug (fixed now) ! :-)

Arvind

ArminRehberger, 2022-09-26:
Hello Arvind
thank you for implementing the new method to draw text onto the screen.
I use your screen driver to show graphic and text.

Kind regards
Armin

vindar, 2022-09-26:
Glad you find the library useful :-) 
I am closing the issue but you can reopen it if you find something wrong with the code.
Thanks for your help !

closed 2022-09-26 by vindar