vindar/ILI9341_T4 #14: ILI9342C Compatibility

windorey, 2023-01-31:
Hi. I have M5Stack Core2 module and I'm creating my own board to fit into it, so I need to use its LCD.
The problem is, the LCD is ILI9342C, and not ILI9341. Does it still work? If not, is there a way to port it?
I have access to the following pins: `RST, BL, MOSI, MISO, SCK, DC, CS`.

windorey, 2023-03-30:
Sorry to bother, but do you have any update?

windorey, 2023-05-22:
> Hi,
> 
> I do not know if the driver can be adapted for use with an ILI9342C screen. I just had a quick look at the datasheet and it seems to be very similar to ILI9341 indeed. So maybe all that is needed is to adjust the init sequence...
> 
> I do not have an ILI9342 screen at hand so I cannot try it right now but I will order one and make some test...

Sorry to bother, but do you have any update?

vindar, 2023-05-27:
HI, 

I finally received a breakout board for an ILI9342C so I will try it as soon as possible (but probably only next week).

vindar, 2023-06-05:
Hello, 

So I (finally) had some time to play with the ILI9342(C) display this weekend. I did not know it uses some weird half-duplex SPI  (i.e. MISO and MOSI are on the same line/PIN)... This make it tricky to adapt the driver to read the screen scanline (which is necessary for vsync)...  But apart from that, it is very similar to ILI9341 so I expect I can port most the driver to this screen. The Vsync feature will probably be disabled (for the first version) but all the other features, such as differential updates should work ok. I will post a first version in a few days.

windorey, 2023-06-05:
> Hello,
>
>
>
> So I (finally) had some time to play with the ILI9342(C) display this weekend. I did not know it uses some weird half-duplex SPI  (i.e. MISO and MOSI are on the same line/PIN)... This [...]

Your specific display seems to have a half duplex data pin, but mine has MOSI and MISO separate from each other. I guess this is a configurational difference? I don't know.

vindar, 2023-06-05:
Hi, 

You have both MOSI and MISO pins ? This is strange because according to the datasheets, the 4-wire SPI interface for ILI9342 (and ILI9342C) consists only of CS, DC, SCK and SDA (which is bi-directional and replaces SDO/SDI). This is different from ILI9341 which can have two separate lines.

Datasheet for ILI9342: https://www.displayfuture.com/Display/datasheet/controller/ILI9342.pdf

Datasheet for ILI9342C:
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/ILI9342C-ILITEK.pdf

Datasheet for ILI9341: 
https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf

In particular, if you look at page 9 [block diagram] of the 3 datasheets, there is only a SDA line for ILI9342(C) whereas there is an additional line SDO for ILI9341. 

Similarly, on page 63 [4 line serial interface], there are two configurations mentioned for 4-wire SPI for ILI9341 (one with SDA and one with both SDI/SDO) but only the SDA configuration is mentioned for the ILI9342 and ILI9342C data sheets...

Are you sure your display is a genuine ILI9342(C) ? I got my breakout board here: https://www.buydisplay.com/serial-spi-2-3-inch-tft-lcd-display-module-with-touch-panel-320x240

the breakout has MISO/MOSI pins but that is for the touchscreen/SD card/flash... The display itself only uses a separate SDA pin.

windorey, 2023-06-05:
> Hi,
>
>
>
> You have both MOSI and MISO pins ? This is strange because according to the datasheets, the 4-wire SPI interface for ILI9342 (and ILI9342C) consists only of CS, DC, SCK and SDA (which is [...]

Hello,

This display comes from the M5Stack Core2 module, and on their official website, it shows that it is indeed an ILI9342C.

https://shop.m5stack.com/products/m5stack-core2-esp32-iot-development-kit
[image]

The schematics also clearly show a MOSI and MISO pin
[image]

This is very unusual. I don't know how it's possible. If I try to use an ILI9341 initialisation on this display (not from this library) it does initialise, but the colours seem washed out and inverted. 

M5Stack seems to provide a data sheet of their display on a hidden link, which is here:
https://m5stack.oss-cn-shenzhen.aliyuncs.com/resource/docs/datasheet/core/ILI9342C-ILITEK.pdf
It loads slowly, it just needs some time, but the link does work.

vindar, 2023-06-05:
Hi, 

Yes, this is very strange indeed. The datasheet you are linking to is the same as the one I referenced above (i.e. without a MISO line)... 

How do you control the screen from the Teensy 4 ? Do you use the Pins on the back of the module ?  I do not know this M5Stack core 2 but it looks like you are supposed to use the onboard ESP32 MCU to drive the display normally.

windorey, 2023-06-05:
> How do you control the screen from the Teensy 4 ? Do you use the Pins on the back of the module ?  I do not know this M5Stack core 2 but it looks like you are supposed to use the onboard ESP32 MCU to drive the display normally. 
> 

Hello,

I haven't controlled it using the Teensy yet, but for now I can test some things using the Arduino framework on ESP32. I have ordered custom breakout PCBs which I will use alongside a new display I got privately from the manufacturer which manufactures these Core2 displays and that will work with the Teensy.

Sorry for any confusion. 

The end goal of my project is to create a Teensy version of the Core2 instead of using the ESP32. 

Also, while you're reading this, is it possible to port this library over to ST7789 too? I apologise for bothering you so much, I will do anything in return for these ports, or if you want I can pay you.

vindar, 2023-06-06:
Hi,

This driver is very specific to Teensy 4 so it is not possible to test it with the embedded ESP32. It looks like the ILI9342 screen I have (without MISO) is different from that on the Core 2 so I see no possibility of testing the driver with the Core 2 right now... Anyway, I am still making a version of the driver for the ILI9342 display I have and I am adding a new mode which disables MISO (and VSync because this feature requires feedback from the screen) which will hopefully also work with the core 2 display when you can try it...

As for porting to ST7789, I will have a look at the datasheet to see if it is doable...

vindar, 2023-06-09:
Hi,

So I finally published a modification of the driver which works with ILI9342 / ILI9342C.  Normally, it should work even without MISO. 

You can find the library there: [https://github.com/vindar/ILI9342_T4](https://github.com/vindar/ILI9342_T4). 

If you can try it, please let me know how it goes...  I am interested to know  if it works with the Core2 screen....

vindar, 2023-06-14:
Hi, 

I am closing the issue here. If there is a problem with the new ILI9342 driver,.you can re open an issue on the new repo: https://github.com/vindar/ILI9342_T4

closed 2023-06-14 by vindar