vindar/ILI9341_T4 #17: Docs for 4bpp v2.3 font converter

sndsgd, 2023-12-21:
I noticed this when trying to use a custom font rendered with `bdf_to_ili9341` from the `ILI9341_t3` repo: https://github.com/vindar/ILI9341_T4/blob/d9bea487d0fa38fc52ffb2bf9e8efc7b827a1cbc/src/ILI9341Driver.cpp#L2145

Can you please add some documentation regarding how you created the font you've included in this library?
https://github.com/vindar/ILI9341_T4/blob/d9bea487d0fa38fc52ffb2bf9e8efc7b827a1cbc/src/font_ILI9341_T4.cpp#L513-L531

My google-fu led me to https://github.com/lvgl/lv_font_conv, but it looks like i'd need to rejigger the output to create `ILI9341_t3_font_t` compatbile code. If there is a tool out there that already does this, I'd much rather use it than re-create it!

sndsgd, 2024-04-12:
looks like [this](https://github.com/projectitis/packedbdf) is what I was looking for.

closed 2024-04-12 by sndsgd

vindar, 2024-04-13:
Hi, sorry for the late reply. Yes,  [https://github.com/projectitis/packedbdf](https://github.com/projectitis/packedbdf) is the creator of the font converter but beware there is a  [bug](https://github.com/projectitis/packedbdf/pull/1) in the script when converting fonts with more than 2 bits resolutions. 

You may also use [my font converter](https://github.com/vindar/tgx-font/tree/main/tools) which is based on projectitis (but hopefully without the bug).