vindar/ILI9341_T4 #20: ILI9341_T4 freezes at _waitUpdateAsyncComplete with tsandmann/freertos-teensy

birdboat00, 2024-08-01:
Hello,

i was using this library with the freertos port by tsandmann (tsandmann/freertos-teensy).

The problem is, it freezes at https://github.com/vindar/ILI9341_T4/blob/d9bea487d0fa38fc52ffb2bf9e8efc7b827a1cbc/src/ILI9341Driver.h#L1433 for some time and then continues, even the SPI clock freezes (atleast the LED stops blinking).

I suspect the problem is the yield(); function.
It works without FreeRTOS.

Is this a problem with the freertos port or with this display driver?

Best regards

vindar, 2024-09-14:
Hi, 

I never used freertos myself so I do not know what is going on but the ILI9341_T4 library make heavy use of interrupts so I am not really surprised that it does not work...  Do you have a simple (complete) example/sketch  that I could try to replicate the issue ? 

Best,

birdboat00, 2024-10-29:
Hey,

sorry for not answering for auch a long time.
For my project I switch from FreeRTOS to ChibiOS and it works there, as it seems, properly.

I will prepare an example when I have time.

Sorry.

Best Regards
Sebastian